daknuett/cryptosuite2 #9: how to store values into an array

mestek86, 2019-02-10:
i need to store the value into a char array instead of printing it but i can not

daknuett, 2019-02-10:
You can take a look at [the documentation here](https://daknuett.github.io/cryptosuite2/usage_arduino.html#simple-usage). You get an `uint8_t` array from `Sha256.result()`. This is the unencoded hash. Note that you **must not** free this array.

You can then encode and store the hash using this code:

```
char encoded[64];
for(int i = 0; i < 64; i+=2)
{
    encoded[i] = "0123456789abcdef"[result[i / 2] >> 4];
    encoded[i + 1] = "0123456789abcdef"[result[i / 2] & 0xf];
}
```

However for most of the applications the unencoded hash will be more handy. You can just cast `uint8_t` to `char`.

closed 2019-02-10 by daknuett

mestek86, 2019-02-10:
thank u very much for ur support,i will try it now

mestek86, 2019-02-10:
#include "sha256.h"
void setup(void)
{char encoded[64];
        Serial.begin(9600);

        // this is actually the RFC4231 4.3 test

        Sha256.init();
        Sha256.print("123");
        uint8_t * result = Sha256.result();

        Serial.println("Expect: 2d711642b726b04401627ca9fbac32f5c8530fb1903cc4db02258717921a4881");
        Serial.print(  "Got   : ");
        for (int i = 0; i < 32; i++) {
                Serial.print("0123456789abcdef"[result[i] >> 4]);
                Serial.print("0123456789abcdef"[result[i] & 0xf]);
                
                for(int i = 0; i < 64; i+=2)
                    {
                      encoded[i] = "0123456789abcdef"[result[i / 2] >> 4];
                      encoded[i + 1] = "0123456789abcdef"[result[i / 2] & 0xf];
                    }
        }
        Serial.print("\n");
                for(int z = 0; z < 64; z++)
                Serial.print(encoded[z]);
}

void loop(void)
{}

this is the code that worked for me
thank u very much
i suggest adding it in ur arduino version as others may need it

daknuett, 2019-02-21:
Feel free to send a pull request. See [our policy on contributing](https://github.com/daknuett/cryptosuite2/blob/master/README.rst#contributing).