daknuett/cryptosuite2 #10: restoring the encoded message

mestek86, 2019-02-21:
i am working in a program that sends the output of sha256 via nRF24l01 i could get the same result to the receiver that i sent from the emitter for the test srting "abc"
i get this result 
"163 0 165 2 50 235 35 0 32 0 32 0 188 225 232 240 16 0 0 0 1 0 32 1 1 0 27 0 243 8 193 0 "
and i need to restore it back to the expected values which are 
"a7816bf8f01cfea414140de5dae2223b00361a396177a9cb410ff61f20015ad"

how to do that as i have read the documentation and cant find a way to do that

daknuett, 2019-02-21:
IMO this is not a cryptosuite2 related problem but it is related to the way you transmit the data. 

However: I think your first set of data is encoded in base 10, while the second set is encoded in base 16.

I don't know where you got the first data from, but I think those are integers you have printed. Then you can convert them to a base 10 string using [`itoa`](http://www.cplusplus.com/reference/cstdlib/itoa/).

Your first set is 32 bytes long, your second set is 31.5 bytes long. Something is really wrong here. Also if you convert the first set of data to hex you will get `a300a50232eb230020002000bce1e8f0100000000100200101001b00f308c100`. It looks like you loose a lot of data while transmitting it.

I will close this issue because it is not related to cryptosuite2 but with your transmitter.

closed 2019-02-21 by daknuett

mestek86, 2019-02-21:
here is the full code of my transmitter

#include "sha256.h"
#include <Arduino.h>
#include "./printf.h"

#include <SPI.h>
//~ #include <TMRh20nRF24L01.h>
//~ #include <TMRh20RF24.h>
#include <nRF24L01.h>
#include <RF24.h>

#define CE_PIN   9
#define CSN_PIN 10

// NOTE: the "LL" at the end of the constant is "LongLong" type
// These are the IDs of each of the slaves
const uint64_t slaveID = 0xE8E8F0F0E1L  ;

RF24 radio(CE_PIN, CSN_PIN); // Create a Radio

unsigned long currentMillis;
unsigned long prevMillis;
unsigned long txIntervalMillis = 1000;
int txVal = 0;
int ackMessg[6];
byte ackMessgLen = 4; 

uint8_t* hash;

void printHash(uint8_t* hash) {
  int i;
  for (i=0; i<32; i++) {
    Serial.print("0123456789abcdef"[hash[i]>>4]);
    Serial.print("0123456789abcdef"[hash[i]&0xf]);
  }
  Serial.println();
}

void setup() {
  printf_begin();
  
  uint32_t a;
  unsigned long ms;
  
  Serial.begin(9600);

  radio.begin();
    radio.setDataRate( RF24_2MBPS );
    radio.enableAckPayload();
    radio.setRetries(3,5); // delay, count

  // SHA tests
  Serial.println("Test: FIPS 180-2 B.1");
  Serial.println("Expect:ba7816bf8f01cfea414140de5dae2223b00361a396177a9cb410ff61f20015ad");
  Serial.print("Result:");
  ms = micros();
  Sha256.init();
  Sha256.print("abc");
  printHash(Sha256.result());
  Serial.print(" Hash took : ");
  Serial.print((micros() - ms));
  Serial.println(" micros");
  Serial.println();
  
  }

void loop() {

  currentMillis = millis();
    if (currentMillis - prevMillis >= txIntervalMillis) {

    radio.openWritingPipe(slaveID); 
    bool rslt,rsltt;
    rslt = radio.write( hash, 32 );
    Serial.print("\nRSLT (1 = success) ");
    Serial.println(rslt);
   
    
    if ( radio.isAckPayloadAvailable() ) {
        radio.read(ackMessg,ackMessgLen);
        Serial.print("Acknowledge received: ");
        Serial.print(ackMessg[0]);
        Serial.print("  ");
        Serial.println(ackMessg[1]);
    }
    prevMillis = millis();
 }

  for(int z=0;z<32;z++){
  Serial.print(hash[z]);
  Serial.print(" ");}

  
  Serial.println();
  
}

daknuett, 2019-02-21:
You did not initialize `hash`.  You have to put a `hash = Sha256.result()` in you setup:

```
hash = Sha256.result();
Serial.println(hash);
```

mestek86, 2019-02-21:
`#include "sha256.h"
#include <Arduino.h>
#include "./printf.h"

uint8_t* hash;

uint8_t copy[32];

void printHash(uint8_t* hash) {
  int i;
  for (i=0; i<32; i++) {
    Serial.print("0123456789abcdef"[hash[i]>>4]);
    Serial.print("0123456789abcdef"[hash[i]&0xf]);
    copy[i]=hash[i];
  }
  Serial.println();
}

void setup() {
  printf_begin();
  
  uint32_t a;
  unsigned long ms;
  
  Serial.begin(9600);

  // SHA tests
  Serial.println("Test: FIPS 180-2 B.1");
  Serial.println("Expect:ba7816bf8f01cfea414140de5dae2223b00361a396177a9cb410ff61f20015ad");
  Serial.print("Result:");
  ms = micros();
  Sha256.init();
  Sha256.print("abc");
  printHash(Sha256.result());
  Serial.print(" Hash took : ");
  Serial.print((micros() - ms));
  Serial.println(" micros");
  Serial.println();
  for(int z=0;z<32;z++)
  Serial.print(copy[z]);
  Serial.println();

}

void loop() {
}`

i have fixed it based on ur words
now i will send it to the nRf24L01
is this right now?

daknuett, 2019-03-21:
Sorry, I didn't see your comment before. What you are doing looks pretty correct. Yes.

mestek86, 2019-03-21:
hello dear

thank u very much for ur reply

yes the code is working correctly