daknuett/cryptosuite2 #15: PlatformIO does not update to the current version

daknuett, 2019-06-14:
Somehow platformIO is still on v0.1.0, although the manifest is updated. @ivankravets do you know what to do here?

ivankravets, 2019-06-14:
It needs 10-15 minutes for our crawler. It's already done https://platformio.org/lib/show/5829/cryptosuite2

daknuett, 2019-06-15:
Well, it was a week. Thank you for your effort.

daknuett, 2023-01-14:
This is just a summary for future reference:

For PlatformIO to detect and pull updates, two conditions must be met:
- The manifest file, i.e., `library.json`, must contain a new version number. Keep in mind that one should also update the Arduino manifest.
- A release must be created. I recommend creating a git tag with the version number and using this tag to publish the release. Also, one should attach the freshly built `sha.zip` for the Arduino users.

I will now close this issue.

closed 2023-01-14 by daknuett