daknuett/cryptosuite2 #16: Improve PlatformIO support

daknuett, 2019-06-15:
PlatformIO has a **ton** of useful options that seem to be helpful (in particular the build options).

`library.json` should be enhanced, preferrably by someone who already has experience with PlatformIO.

stefanfisk, 2020-02-25:
The library does not build at all for me. `/sha/test` seem to be included when building, and theu use system includes rather than local includes for `sha1.h` / `sha256.h`. Changing the includes to relative local makes the library build successfully.

```
Processing sonoff_sv (platform: espressif8266; board: sonoff_sv; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/sonoff_sv.html
PLATFORM: Espressif 8266 2.3.3 > Sonoff SV
HARDWARE: ESP8266 160MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.191122 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ off
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <cryptosuite2> 0.2.4
Building in release mode
Compiling .pio/build/sonoff_sv/src/main.cpp.o
Compiling .pio/build/sonoff_sv/lib74f/cryptosuite2_ID5829/sha/test/test_sha1.c.o
Compiling .pio/build/sonoff_sv/lib74f/cryptosuite2_ID5829/sha/test/test_sha256.c.o
Archiving .pio/build/sonoff_sv/libFrameworkArduinoVariant.a
cc1: warning: command line option '-fpermissive' is valid for C++/ObjC++ but not for C [enabled by default]
cc1: warning: command line option '-fpermissive' is valid for C++/ObjC++ but not for C [enabled by default]
.pio/libdeps/sonoff_sv/cryptosuite2_ID5829/sha/test/test_sha1.c:17:23: fatal error: sha1/sha1.h: No such file or directory
 #include <sha1/sha1.h>
                       ^
compilation terminated.
.pio/libdeps/sonoff_sv/cryptosuite2_ID5829/sha/test/test_sha256.c:17:27: fatal error: sha256/sha256.h: No such file or directory
 #include <sha256/sha256.h>
                           ^
compilation terminated.
Indexing .pio/build/sonoff_sv/libFrameworkArduinoVariant.a
Compiling .pio/build/sonoff_sv/FrameworkArduino/Esp-frag.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Esp-version.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/FS.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/FSnoop.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Print.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Schedule.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/StackThunk.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/Updater.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/WString.cpp.o
Compiling .pio/build/sonoff_sv/FrameworkArduino/abi.cpp.o
*** [.pio/build/sonoff_sv/lib74f/cryptosuite2_ID5829/sha/test/test_sha256.c.o] Error 1
*** [.pio/build/sonoff_sv/lib74f/cryptosuite2_ID5829/sha/test/test_sha1.c.o] Error 1
Compiling .pio/build/sonoff_sv/FrameworkArduino/base64.cpp.o
================================================================= [FAILED] Took 0.86 seconds =================================================================
```

daknuett, 2020-02-25:
@stefanfisk Thank you very much for reporting this. I fixed the includes in 0811b0bed3e8f53091ae115a9fdc19b940a3e145. 
I don't remember why I decided to leave just two "test" files there, now they have some infrastructure around them and everything should work finely.

Maybe it is a mistake that PlatformIO compiles the test files (does it also try to link them?). TBH I know very little about PIO.

stefanfisk, 2020-02-27:
Thanks, now it builds as-is (when using `lib_deps = https://github.com/daknuett/cryptosuite2`).

**DISCLAIMER**: My current project is my first time trying out embedded stuff at all, and I don't have much experience with the inner workings of C/C++ build systems, so please don't assume that I know what I'm talking about 😸

When checking with `nm` I can see that the test files are being linked into a `libcryptosuite2.a` during the build. I cannot however find any traces of the test file symbols in the resulting `firmware.elf`.

daknuett, 2020-02-27:
Well the PlatformIO build system is way more complicated than writing a Makefile. They somehow support more frameworks (like Arduino) as well; I will have to figure out how to make the library work on other frameworks as well. Defining `#define SHAx_DISABLE_WRAPPER` in `config.h` should make it work on any framework but I have not tested it...

I hope that people facing problems will just report them here so I can fix them.

commit 81396015 (2021-12-17): Merge pull request #21 from StarGate01/fix/platformio-tests

Remove the test sketches from the regular build on PlatformIO. Fixes #16.
closed 2021-12-17 by daknuett