daknuett/cryptosuite2 #17: Converting result to string

revoxhere, 2020-05-01:
Hello, I'm the main developer of [Duino-Coin](https://revoxhere.github.io/duino-coin/), we used to use [Hash.h](https://github.com/esp8266/Arduino/blob/master/libraries/Hash/src/Hash.h) library for ESP8266 boards but it doesn't work with Arduino boards. After long search, I've found this great sha1 implementation which works on both platforms. 
I need help though, this a snippet of the current [Arduino code](https://github.com/revoxhere/duino-coin/blob/master/Arduino_Code/Arduino_Code.ino): 
```C++
void DUCOS1A() {
  if (Serial.available() > 0) { // Wait for serial to become available
    String hash = Serial.readStringUntil('\n'); // Read hash
    String job = Serial.readStringUntil('\n'); // Read job
    unsigned int diff = Serial.readStringUntil('\n').toInt(); // Read difficulty
    for (unsigned int iJob = 0; iJob < diff * 100 + 1; iJob++) { // Difficulty loop
      yield(); // Let Arduino/ESP do background tasks - else watchdog will trigger
      String result = sha1(String(hash) + String(iJob)); // Hash previous block hash and current iJob
      if (result == job) { // If result is found
        Serial.println(iJob); // Send result back to Arduino Miner
        digitalWrite(LED_BUILTIN, HIGH);   // Turn on built-in led
        delay(50); // Wait a bit
        digitalWrite(LED_BUILTIN, LOW); // Turn off built-in led
        break; // Stop and wait for more work
      }
    }
  }
}
```
I want to "transform" it to cryptosuite2 library, I don't know how to get the sha1 result as a string for comparison though. 
I saw in [another issue](https://github.com/daknuett/cryptosuite2/issues?q=is%3Aissue+is%3Aclosed) there was someone converting it to array but I don't have enough experience with C++ stuff to make it work in my case.
Can someone help with this?
Thanks in advance,
Robert "revox" Piotrowski

daknuett, 2020-05-01:
I am not certain that the code does what you think it does. But that is a separated problem.

You can convert the 32 byte result to a 64 byte `char*` buffer by using

```
char buffer[64];
for(int i = 0; i < 32; i++)
{
    buffer[2*i] = "0123456789abcdef"[hash_bytes[i]>>4];
    bufffer[2*i + 1] = "0123456789abcdef"[hash_bytes[i]&0xf];
}
```

and then convert this buffer to a `String` object.

Make sure to initialize and finalize the hasher properly, i.e. in every loop start with
`Sha1.init()` and then get your result using `uint8_t * hash_bytes = Sha1.result()`. This result is a reference to the hash, so don't free it.

I am closing this issue, you can still comment and ask more.

closed 2020-05-01 by daknuett

revoxhere, 2020-05-01:
Thank You very much! 
With your help I managed to find a solution :smiley:
I will include your name in the next release!

revoxhere, 2022-05-06:
This keeps me up at night. What was wrong with my question at that time @daknuett ? 🤔 

> I am not certain that the code does what you think it does. But that is a separated problem.

daknuett, 2022-05-06:
I do not remember why I made that comment. You are inverting some hashes, right? I probably misunderstood the task. Mea culpa.