daknuett/cryptosuite2 #20: Not working on Arduino DUE

erdavci, 2021-05-18:
Arduino: 1.8.13 (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from c:\..\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\stdlib.h:11:0,

from C:\..\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:24,

from sketch\sha1.ino.cpp:1: c:\..\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\sys\unistd.h:118:9: error: declaration of C function 'int link(const char*, const char*)' conflicts with

int     _EXFUN(link, (const char *__path1, const char *__path2 ));
^
In file included from C:\..\Arduino15\packages\arduino\hardware\sam\1.6.12\variants\arduino_due_x/variant.h:50:0,

from C:\..\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:201,

from sketch\sha1.ino.cpp:1: C:\..\Arduino15\packages\arduino\hardware\sam\1.6.12\cores\arduino/syscalls.h:43:12: error: previous declaration 'int link(char*, char*)' here

extern int link( char *cOld, char *cNew ) ;
^
exit status 1

Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

daknuett, 2021-05-18:
This appears to be a conflict between `unistd.h` and `arduino/syscalls.h`. I think opening an issue at the Arduino repository would be useful.

I will try to remove `unistd.h` from the library. I think I used it to get `size_t` which is a pain to obtain (particularly when mixing `C` and `C++`).

daknuett, 2021-09-22:
I doubt that I will fix this problem in the current messy code base.

When I have time to do so I will clean up the code and fix this issue. Until then contributions are welcome but I will not fix it myself.

closed 2022-01-19 by erdavci