daknuett/cryptosuite2 #25: daknuett/cryptosuite2@^0.2.5 from PlatformIO conflicts with Arduino UNO

snakeninny, 2023-01-12:
Hi,

I have an Arduino UNO board and am testing with PlatformIO. The code is pretty straightforward:
```
#include <Arduino.h>

void setup()
{
  Serial.begin(9600);
  Serial.println("Hello World");
}

void loop()
{
}
```

platformio.ini
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

Upload then I get:
```
 *  Executing task: platformio device monitor 

--- Terminal on /dev/cu.wchusbserial1120 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Hello World
```

But with the library `daknuett/cryptosuite2@^0.2.5` in platformio.ini like this:
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps = 
	daknuett/cryptosuite2@^0.2.5
```

No code seems to be running and the serial monitor outputs no "Hello World":
```
 *  Executing task: platformio device monitor 

--- Terminal on /dev/cu.wchusbserial1120 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
```

Any ideas? Thank you!

daknuett, 2023-01-12:
Unfortunately, PlatformIO support is experimental at best. I currently do not have an Arduino to test the behavior but 
I can reproduce the problem using simavr. This is indeed a very strange bug.

I am not quite certain what could cause this problem.

snakeninny, 2023-01-13:
Thanks!
How about you update to 0.2.6 on [platformIO registry](https://registry.platformio.org/libraries/daknuett/cryptosuite2) then let's see if the bug persists?

daknuett, 2023-01-13:
This is a good idea, however, does PlatformIO crawl for new versions and there is nothing we can do to bump the version. See also Issue  #15 .

I now tried to use the github version explicitly, i.e., 

    lib_deps = https://github.com/daknuett/cryptosuite2

using the current github version does work at least on simAVR.

I will see what I can do about the PlatformIO problems.

ivankravets, 2023-01-13:
@daknuett , the [0.2.6](https://github.com/daknuett/cryptosuite2/blob/v0.2.6/library.json) release is broken. The manifest points to the 0.2.5

snakeninny, 2023-01-14:
> ```
> lib_deps = https://github.com/daknuett/cryptosuite2
> ```

FYI, Arduino UNO works for 0.2.6.

daknuett, 2023-01-14:
I released v0.2.7. This should trigger an update on PlatformIO (at least if I understood correctly) and solve this issue.
I will keep the issue open until you confirmed that it works.

ivankravets, 2023-01-14:
Thanks, @daknuett. I see that the new version has been successfully published 
https://registry.platformio.org/libraries/daknuett/cryptosuite2

daknuett, 2023-01-14:
Thank you for your kind help, @ivankravets .

I will close both issues.

closed 2023-01-14 by daknuett

snakeninny, 2023-01-15:
I have tested Arduino UNO and NodeMCU, both worked like a charm. Thank you!

sadatusa, 2023-10-29:
I am still facing same problem, with all update. Please help.
Thanks
Sadat

daknuett, 2023-10-30:
@sadatusa can you provide some more information? What platform are you using? How are you pulling cryptosuite2? What version are you using?

sadatusa, 2023-10-30:
Dear, Thanks for your quick response. The issue was in my source codes but behavior was similar as describe here. Which makes me confused, Sorry for disturbing you.Thanks Sadat